freebsd/crochet #297: Building releng/13.3 for BeagleBone Black

ppaeps, 2024-05-20:
I have a crusty old BeagleBone Black with a very old 12-CURRENT crochet build on it.  I realise eventually this board will need to die (TI support is gone in stable/14), but I'd like to get a final releng/13.3 on it, to squeeze a couple more years of life out of it.

```
board_setup BeagleBone
option ImageSize 3900mb
option User philip
FREEBSD_SRC=${TOPDIR}/../releng-133
```

Building with `sudo ./crochet.sh -c bbb.sh` falls over with:

```
Starting at Tue May 21 07:51:21 HKT 2024
Loading configuration from bbb.sh
Board: BeagleBone
Option: ImageSize 3900mb
Option: User philip
Source version is: unknown
Building FreeBSD version: 13.3
Image name is:
    /home/philip/freebsd/crochet/work/FreeBSD-armv6-13-GENERIC-BeagleBone.img
Building FreeBSD version: 13.3
Object files are at: /home/philip/freebsd/obj/home/philip/freebsd/releng-133
Found suitable FreeBSD source tree in:
    /home/philip/freebsd/crochet/../releng-133
Found U-Boot port in:
    /usr/local/share/u-boot/u-boot-beaglebone
Using FreeBSD armv6 world from previous build
Using FreeBSD armv6-GENERIC kernel from previous build
Building FreeBSD armv6-GENERIC ubldr at Tue May 21 07:51:22 HKT 2024
    (Logging to /home/philip/freebsd/crochet/work/ubldr-armv6-GENERIC/_.ubldr.armv6-GENERIC.build.log)
cd: arm/uboot: No such file or directory
```

I tried doing a clean build.  Also falls over here.

I guess arm/uboot moved somewhere between the last time someone loved crochet and releng/13.3.

Anyone have a clue to save me some time grepping please? :)

christopher-bowman, 2024-05-21:
There are a whole bunch of U-boot ports, some of which were even created, and then depreciated. I myself maintain an out of tree, you u-boot port.  Most ports, I think, rely on a master port. Mine is just a few patches on top of that.  You might check out the last u-boot port for your board and use that or try using the latest u-boot with the existing master port framework.  I’m not quite sure how you would patch this into crochet, however.I did use crochet for a while, but then I switched over to using the native make targets with cross compiling. It’s really not that hard and I can help you if you like.Christopher

ppaeps, 2024-05-21:
I'm not married to crochet. :)

If you have another way of building an image I can dd to my /dev/mmcsd1, I'm all ears.

Many thanks!